dorimanx/exfat-nofuse #56: Kernel panic on android mtk 3.4.67

Niorix, 2015-01-15:
Module build and insmod work propertly, but on mount device device reboots
root@U65GT:/system/lib/modules # uname -a
Linux localhost 3.4.67 #1 SMP PREEMPT Tue Nov 11 14:26:57 CST 2014 armv7l GNU/Linux
root@U65GT:/system/lib/modules # cat /proc/filesystems
nodev   sysfs
nodev   rootfs
nodev   bdev
nodev   proc
nodev   cgroup
nodev   tmpfs
nodev   debugfs
nodev   sockfs
nodev   pipefs
nodev   anon_inodefs
nodev   devpts
        ext3
        ext4
nodev   ramfs
        vfat
        msdos
        iso9660
        fuseblk
nodev   fuse
nodev   fusectl
nodev   selinuxfs
nodev   mtd_inodefs
nodev   functionfs
root@U65GT:/system/lib/modules # insmod exfat.ko ; dmesg | grep -i exfat
<6>[  357.474296] 1[5567:insmod]exFAT: Version 1.2.9
root@U65GT:/system/lib/modules # cat /proc/filesystems
nodev   sysfs
nodev   rootfs
nodev   bdev
nodev   proc
nodev   cgroup
nodev   tmpfs
nodev   debugfs
nodev   sockfs
nodev   pipefs
nodev   anon_inodefs
nodev   devpts
        ext3
        ext4
nodev   ramfs
        vfat
        msdos
        iso9660
        fuseblk
nodev   fuse
nodev   fusectl
nodev   selinuxfs
nodev   mtd_inodefs
nodev   functionfs
        exfat
mount -t exfat /dev/block/vold/179\:97 /mnt/secure/staging/

But on mount device device reboots
root@U65GT:/storage/sdcard0 # cat /proc/last_kmsg
ram console header, hw_status: 2, fiq step 40.
bin log 0.
ng                 : 0
[  650.248919] \(2\)[13186:mount]  .rt_throttled                  : 0
[  650.248931] \(2\)[13186:mount]  .rt_time                       : 0.000000
[  650.248944] \(2\)[13186:mount]  .rt_runtime                    : 700.000000
[  650.248957] \(2\)[13186:mount]
[  650.248961] \(2\)[13186:mount]rt_rq[7]:/apps
[  650.248973] \(2\)[13186:mount]  .rt_nr_running                 : 0
[  650.248985] \(2\)[13186:mount]  .rt_throttled                  : 0
[  650.248997] \(2\)[13186:mount]  .rt_time                       : 0.000000
[  650.249010] \(2\)[13186:mount]  .rt_runtime                    : 800.000000
[... 141 lines cut ...]
[  404.779626] \(1\)[6286:mount][\<c0008374\>] \(do_DataAbort+0x0/0x110\) from [\<c000e258\>] \(__dabt_svc+0x38/0x60\)
[  404.779642] \(1\)[6286:mount]Exception stack\(0xcbdfdd70 to 0xcbdfddb8\)
[  404.779658] \(1\)[6286:mount]dd60:                                     ffffffff ca434100 00000001 00000000
[  404.779678] \(1\)[6286:mount]dd80: ca4340ec bf00fa3c c5babc00 ca430000 00000001 00000003 00000000 cbdfde2c
[  404.779696] \(1\)[6286:mount]dda0: 00000000 cbdfddb8 c00e8434 bf00925c 60070013 ffffffff
[  404.779709] \(1\)[6286:mount] r8:00000001 r7:cbdfdda4 r6:ffffffff r5:60070013 r4:bf00925c
[  404.779782] \(1\)[6286:mount][\<bf0091a0\>] \(exfat_fill_super+0x0/0x4c0 [exfat]\) from [\<c011acbc\>] \(mount_bdev+0x174/0x1dc\)
[  404.779827] \(1\)[6286:mount][\<c011ab48\>] \(mount_bdev+0x0/0x1dc\) from [\<bf009198\>] \(exfat_fs_mount+0x20/0x28 [exfat]\)
[  404.779871] \(1\)[6286:mount][\<bf009178\>] \(exfat_fs_mount+0x0/0x28 [exfat]\) from [\<c011b784\>] \(mount_fs+0x4c/0x194\)
[  404.779899] \(1\)[6286:mount][\<c011b738\>] \(mount_fs+0x0/0x194\) from [\<c0133b1c\>] \(vfs_kern_mount+0x50/0xe8\)
[  404.779925] \(1\)[6286:mount][\<c0133acc\>] \(vfs_kern_mount+0x0/0xe8\) from [\<c0134724\>] \(do_kern_mount+0x38/0xfc\)
[  404.779940] \(1\)[6286:mount] r8:d7d8f440 r7:00008000 r6:d7d8f7c0 r5:bf00fa44 r4:00000000
[  404.779972] \(1\)[6286:mount]r3:00000000
[  404.779999] \(1\)[6286:mount][\<c01346ec\>] \(do_kern_mount+0x0/0xfc\) from [\<c0135fc4\>] \(do_mount+0x334/0x854\)
[  404.780014] \(1\)[6286:mount] r8:d7d8f7c0 r7:00000020 r6:00000000 r5:00000000 r4:00008000
[  404.780045] \(1\)[6286:mount]r3:00000000
[  404.780072] \(1\)[6286:mount][\<c0135c90\>] \(do_mount+0x0/0x854\) from [\<c0136640\>] \(sys_mount+0x8c/0xc0\)
[  404.780099] \(1\)[6286:mount][\<c01365b4\>] \(sys_mount+0x0/0xc0\) from [\<c000e800\>] \(ret_fast_syscall+0x0/0x30\)
[  404.780113] \(1\)[6286:mount] r7:00000015 r6:00000000 r5:00000000 r4:00000000
[  404.780141] \(1\)[6286:mount]machine_shutdown: done
[  404.780151] \(1\)[6286:mount]Reboot:machine restart...
[  404.782136] \(1\)[6286:mount]arch_reset: cmd = NULL
[  404.782373] \(1\)[6286:mount]wdt_arch_reset called@Kernel mode =
[  404.782857] \(1\)[6286:mount]wdt_arch_reset called MTK_WDT_MODE =5c
[  404.783323] \(1\)[6286:mount]wdt_arch_reset called end  MTK_WDT_MODE =22000014

dorimanx, 2015-11-10:
was made for samsung/LG/Asus/Nexus. MTK is not supported.

closed 2015-11-10 by dorimanx

mcirsta, 2015-11-10:
Lol, a file system driver is not made for a brand or a specific SOC. Of course if the error is indeed caused by something in the kernel code for that SOC but ....

mlochd, 2016-01-24:
Hi, did anybody get exfat working on an mtk kernel?