dorimanx/exfat-nofuse #58: Error when compiling on Debian Wheezy amd64: incompatible type for argument 3 of ‘utf8s_to_utf16s’

benjaminbrent, 2015-01-20:
I am seeing the following errors when compiling on Debian Wheezy amd64:

```
$ make
/usr/src/linux-headers-3.2.0-4-common/scripts/Makefile.build:44: /usr/src/linux-headers-3.2.0-4-common/scripts/basic/Makefile: No such file or directory
make[4]: *** No rule to make target `/usr/src/linux-headers-3.2.0-4-common/scripts/basic/Makefile'.  Stop.
make[3]: *** [scripts_basic] Error 2
make -C /lib/modules/3.2.0-4-amd64/build M=/home/benjamin/Repos/exfat-nofuse modules
make[1]: Entering directory `/usr/src/linux-headers-3.2.0-4-amd64'
  CC [M]  /home/benjamin/Repos/exfat-nofuse/exfat_nls.o
/home/benjamin/Repos/exfat-nofuse/exfat_nls.c: In function ‘nls_cstring_to_uniname’:
/home/benjamin/Repos/exfat-nofuse/exfat_nls.c:357:3: error: incompatible type for argument 3 of ‘utf8s_to_utf16s’
/usr/src/linux-headers-3.2.0-4-common/include/linux/nls.h:55:12: note: expected ‘enum utf16_endian’ but argument is of type ‘u16 *’
/home/benjamin/Repos/exfat-nofuse/exfat_nls.c:357:3: error: too few arguments to function ‘utf8s_to_utf16s’
/usr/src/linux-headers-3.2.0-4-common/include/linux/nls.h:55:12: note: declared here
make[4]: *** [/home/benjamin/Repos/exfat-nofuse/exfat_nls.o] Error 1
make[3]: *** [_module_/home/benjamin/Repos/exfat-nofuse] Error 2
make[2]: *** [sub-make] Error 2
make[1]: *** [all] Error 2
make[1]: Leaving directory `/usr/src/linux-headers-3.2.0-4-amd64'
make: *** [all] Error 2
```

x545, 2015-01-20:
kernel 3.2.0 is a little bit old...

benjaminbrent, 2015-01-21:
Shouldn't it still be supported, considering that's the kernel supported by Debian stable, a major distribution on which many other distributions are based?

asida, 2015-03-01:
I've upg my Deb7 amd64 from kernel 3.2 to 3.16 to run/test exFat...
No one problem with Deb after upg (vbox asks for kernel module rebuild, success)
But still can't build exFat module with error:
root@deb64:/home/asida/exfat-nofuse# make
/usr/src/linux-headers-3.16.0-0.bpo.4-common/scripts/Makefile.build:44: /usr/src/linux-headers-3.16.0-0.bpo.4-common/scripts/basic/Makefile: no such file or directory
make[4]: *** No rule to make target `/usr/src/linux-headers-3.16.0-0.bpo.4-common/scripts/basic/Makefile'.  Stop.
make[3]: *** [scripts_basic] Error 2
make -C /lib/modules/3.16.0-0.bpo.4-amd64/build M=/home/asida/exfat-nofuse modules
make[1]: Entering directory `/usr/src/linux-headers-3.16.0-0.bpo.4-amd64'
make[1]: Entering directory `/usr/src/linux-headers-3.16.0-0.bpo.4-amd64'
  Building modules, stage 2.
  MODPOST 1 modules
make[1]: Leaving directory `/usr/src/linux-headers-3.16.0-0.bpo.4-amd64'

can I please get some idiots guide to get this work? I've tried search www ....installed all suggested packages...but none I found helped :-/

also did someone test exFat-nofuse performance ? my ssd is 192MB/s on ext3/4 and I can use exFat-nofuse only when similar perf, otherwise it is pointless...

many thx for any feedback

dorimanx, 2015-03-01:
as far as i know. this driver was written for 32Bit linux.
so it's will fail to compile under 64bit source.

big-sheep, 2015-04-13:
Hi @benjaminbrent,
I've compiled successfully by removing the incompatible calls in the source code.
I'm using kernel 3.2.0-4-amd64 #1 SMP Debian 3.2.65-1+deb7u2
GNU/Linux).
Just open exfat_nls.c and go to line 356:

```
#if LINUX_VERSION_CODE < KERNEL_VERSION(3,3,0)
                i = utf8s_to_utf16s(p_cstring, MAX_NAME_LENGTH, uniname);
#else
                i = utf8s_to_utf16s(p_cstring, MAX_NAME_LENGTH, UTF16_HOST_ENDIAN, uniname, MAX_NAME_LENGTH);
#endif
```

Just replace the above code with:
```
                i = utf8s_to_utf16s(p_cstring, MAX_NAME_LENGTH, UTF16_HOST_ENDIAN, uniname, MAX_NAME_LENGTH);
```

dmesg results:
```
[498438.929630] [EXFAT] trying to mount...
[498439.084837] [EXFAT] mounted successfully
[498449.480398] [EXFAT] trying to unmount...
[498449.480455] [EXFAT] unmounted successfully
```

dorimanx, 2015-04-13:
Fix is OK. but it's not something i can merge. your source is UPDATED... and real 3.2.y should not have that need for change.
if you wish to offer update that solve the problem, and NOT mess other builds. i will merge it after review.
Any way thanks for info. case closed.

closed 2015-04-13 by dorimanx