dorimanx/exfat-nofuse #69: Won't build with 4.1 + vfs-next

LynxChaus, 2015-07-04:
Fails to build with vfs-next tree due to https://git.kernel.org/cgit/linux/kernel/git/viro/vfs.git/commit/?h=for-next&id=680baacbca69d18a6d7315374ad83d05ac9c0977 commit

make -C /lib/modules/4.1.0-999-generic/build M=/var/lib/dkms/exfat/1.2.9-20150703/build modules
make[1]: Entering directory '/usr/src/linux-headers-4.1.0-999-generic'
  CC [M]  /var/lib/dkms/exfat/1.2.9-20150703/build/exfat_core.o
  CC [M]  /var/lib/dkms/exfat/1.2.9-20150703/build/exfat_super.o
/var/lib/dkms/exfat/1.2.9-20150703/build/exfat_super.c: In function ‘exfat_follow_link’:
/var/lib/dkms/exfat/1.2.9-20150703/build/exfat_super.c:1314:2: error: implicit declaration of function ‘nd_set_link’ [-Werror=implicit-function-declaration]
  nd_set_link(nd, (char *)(ei->target));
  ^
/var/lib/dkms/exfat/1.2.9-20150703/build/exfat_super.c: At top level:
/var/lib/dkms/exfat/1.2.9-20150703/build/exfat_super.c:1320:2: warning: initialization from incompatible pointer type
  .follow_link = exfat_follow_link,
  ^
/var/lib/dkms/exfat/1.2.9-20150703/build/exfat_super.c:1320:2: warning: (near initialization for ‘exfat_symlink_inode_operations.follow_link’)
/var/lib/dkms/exfat/1.2.9-20150703/build/exfat_super.c:1679:2: warning: initialization from incompatible pointer type
  .direct_IO   = exfat_direct_IO,
  ^
/var/lib/dkms/exfat/1.2.9-20150703/build/exfat_super.c:1679:2: warning: (near initialization for ‘exfat_aops.direct_IO’)
cc1: some warnings being treated as errors
scripts/Makefile.build:258: recipe for target '/var/lib/dkms/exfat/1.2.9-20150703/build/exfat_super.o' failed
make[2]: *** [/var/lib/dkms/exfat/1.2.9-20150703/build/exfat_super.o] Error 1
Makefile:1380: recipe for target '_module_/var/lib/dkms/exfat/1.2.9-20150703/build' failed
make[1]: *** [_module_/var/lib/dkms/exfat/1.2.9-20150703/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.1.0-999-generic'
Makefile:36: recipe for target 'all' failed
make: *** [all] Error 2

dorimanx, 2015-07-05:
This driver based on 3.0.11 :-) after many mods its support higher builds.
But i guess you will have to adapt to 4.1
I have no idea how to, my knowledge is limited to 3.4.x up to 3.8.x.

LynxChaus, 2015-07-06:
Changes minimal, but reading exfat_direct_IO routine is ugly.
```
--- a/exfat_super.c
+++ b/exfat_super.c
@@ -1307,13 +1307,20 @@ const struct inode_operations exfat_dir_inode_operations = {
 /*======================================================================*/
 /*  File Operations                                                     */
 /*======================================================================*/
-
+#if LINUX_VERSION_CODE > KERNEL_VERSION(4,1,0)
+static const char *exfat_follow_link(struct dentry *dentry, void **cookie)
+{
+       struct exfat_inode_info *ei = EXFAT_I(dentry->d_inode);
+       return *cookie = (char *)(ei->target);
+}
+#else
 static void *exfat_follow_link(struct dentry *dentry, struct nameidata *nd)
 {
        struct exfat_inode_info *ei = EXFAT_I(dentry->d_inode);
        nd_set_link(nd, (char *)(ei->target));
        return NULL;
 }
+#endif
 
 const struct inode_operations exfat_symlink_inode_operations = {
        .readlink    = generic_readlink,
@@ -1591,9 +1598,12 @@ static ssize_t exfat_direct_IO(int rw, struct kiocb *iocb,
                                           const struct iovec *iov,
                                           loff_t offset, unsigned long nr_segs)
 #endif
-#else
+#elif LINUX_VERSION_CODE < KERNEL_VERSION(4,2,0)
 static ssize_t exfat_direct_IO(int rw, struct kiocb *iocb,
                                           struct iov_iter *iter, loff_t offset)
+#else /* >= 4.2.x */
+static ssize_t exfat_direct_IO(struct kiocb *iocb,
+                                          struct iov_iter *iter, loff_t offset)
 #endif
 {
        struct inode *inode = iocb->ki_filp->f_mapping->host;
@@ -1601,7 +1611,12 @@ static ssize_t exfat_direct_IO(int rw, struct kiocb *iocb,
        struct address_space *mapping = iocb->ki_filp->f_mapping;
 #endif
        ssize_t ret;
+#if LINUX_VERSION_CODE >= KERNEL_VERSION(4,2,0)
+       int rw;
 
+       rw = iov_iter_rw(iter);
+#endif
+       
        if (rw == WRITE) {
 #if LINUX_VERSION_CODE < KERNEL_VERSION(3,16,0)
 #ifdef CONFIG_AIO_OPTIMIZATION
```

dorimanx, 2015-07-11:
I will find time to add this mod in,  or send me pull request that fix for 4.1 and not brake others :-) and i will merge with credits to you.

LynxChaus, 2015-07-12:
Pull request generated. compiled and tested with 4.2.0-rc1-g59c3cb5.

dorimanx, 2015-11-10:
Thanks. problem resolved.

closed 2015-11-10 by dorimanx