dorimanx/exfat-nofuse #71: It makes corruptions to the filesystem

wisecrick, 2015-08-10:
Even I made little modifications to the filesystem (eg: move one file from one directory to another), the windows chkdsk utility will report the filesystem has some errors.

C:\Windows\system32>chkdsk /f e:
The type of the file system is exFAT.
Volume Serial Number is 34BF-9407
Windows is verifying files and folders...
Volume label is DATA.
Corruption was found while examining files in directory \soft\Adobe\............

Windows has made corrections to the file system.

lqs, 2015-08-17:
try to mount with `-o namecase=0`

ismail, 2015-08-19:
exfat has no case sensitivity according to https://msdn.microsoft.com/en-us/library/windows/desktop/ee681827(v=vs.85).aspx#functionality , I wonder why the code has an option for this.

wisecrick, 2015-08-19:
Well, it works at present, and I also wonder what happens if I don't specify this option ( and why windows chkdsk will report the errors).

lgrootnoob, 2015-11-09:
What about on android. we have auto detection of sdcards. will it be okay to put this flag in the fstab and have the other filesystems ignore it? like if ext4 sdcard was inserted it would just ignore the non existent flag and still mount?

dorimanx, 2015-11-10:
fixed.
https://github.com/dorimanx/exfat-nofuse/issues/71

exfat should not be casesensitive. it's win partition.

closed 2015-11-10 by dorimanx