dorimanx/exfat-nofuse #74: No dokumentation for CONFIG_EXFAT_DELAYED_SYNC

V10lator, 2015-10-09:
I want to use this code on a mobile phone but fear risk of data loss or corruption when not unmounting correctly (lazy pulling out sd card). I can't do "mount -o flush", so I looked at the codes a bit and found CONFIG_EXFAT_DELAYED_SYNC instead but this variable is not commented at any place.

Now from the name and with common sense I would say it delays a sync, so it's the opposite of what I want but when I look at the codes I see this

#ifdef CONFIG_EXFAT_DELAYED_SYNC
	fs_sync(sb, 0);
	fs_set_vol_flags(sb, VOL_CLEAN);
#endif

which looks as it is exactly what I want.

Now I'm not that experienced and don't want others to risk their data, so could you please tell me what the variable does exactly, at best comment the variable (at exfat-config.h) ?

Bonus question: To activate the code (the ifdef condition gets true) I have to set the variable from 0 to 1, right?

lgrootnoob, 2015-11-09:
I would like to know this too. Of course, the dirty way is to just look at write speeds when it's enabled versus not enabled. The faster write speeds would be delayed sync. if this option makes it slower i would guess it actually syncs.

dorimanx, 2015-11-10:
seems right. ask Samsung. they never told us that its can be used with lazy sd removal.
as all or most... samsung devices dont let you remove sd without removing battery first.
test and update the issue or close it.

closed 2016-11-21 by dorimanx