dorimanx/exfat-nofuse #75: 3.4.110 clear_inode not existing

lgrootnoob, 2015-11-09:
Code is here 
#if LINUX_VERSION_CODE < KERNEL_VERSION(3,4,80)
 	end_writeback(inode);
else
clear_inode(inode);
Shouldn't the arrow be the other way around? my compilation fails if it's as above. but if i change it to this: 
#if LINUX_VERSION_CODE > KERNEL_VERSION(3,4,80)
it compiles.
What gives? here is my source https://github.com/lgrootnoob/android_kernel_asus_grouper
Thanks!

dorimanx, 2015-11-10:
its is correct.
you have made updates to FS. and changed the end_writeback to clear_inode
this change was added at 3.4.80 patch.
dont open issues that belong to your build only.

closed 2015-11-10 by dorimanx

lgrootnoob, 2016-06-27:
so what happened here then?  im confused now man https://github.com/dorimanx/exfat-nofuse/commit/5156845e4f391ada85659dc8ba8d95bd35af3653

lgrootnoob, 2016-06-27:
i remember picking that commit from vanilla kernel.org git branches to switch the function's name to clear_inode. odd

dorimanx, 2016-06-28:
Hmm... kernel now days is not what used to be.
3.4.y has parts from 4.5.y and lower in many cases to fix some ugly bugs.
some choose to switch to end_writeback some clear_inode

THIS IS THE SAME FUNCTION. DOING SAME THING. different name.