dorimanx/exfat-nofuse #79: Support for Linux 2.6.30

kirpal06391, 2015-12-11:
Is Linux 2.6.30 supported. I'm trying to compile it for my Dlink router which has an archaic linux 2.6.30 kernel. Apparently during compilation I found many functions which were implicitly declared, on further research I found all those functions to be present in Linux 2.6.31 but not in 2.6.30. Any help will be appreciated.

dorimanx, 2015-12-12:
Driver was originally made for 3.0.10 so older kernels will have problems yo compile and run it even with corrected functions.
as no one tested this before.
If you will be able to correct all needed and its will work, send pull or patch.

dorimanx, 2016-11-21:
well no one interested. i will close the case here....

closed 2016-11-21 by dorimanx