dorimanx/exfat-nofuse #80: pulling drive, then umount will freak out driver

dannydulai, 2016-01-14:
here is the pull from USB:
[   52.987205] usb 2-1: USB disconnect, device number 2

here is mdev trying to umount the mountpoint because the drive disappeared:

[   53.001854] [EXFAT] trying to unmount...
[   53.003880] ------------[ cut here ]------------
[   53.005149] WARNING: CPU: 3 PID: 344 at fs/exfat/exfat_blkdev.c:124 bdev_read+0xc9/0xd0()
[   53.006425] [EXFAT] No bh, device seems wrong or to be ejected.
[   53.007680] Modules linked in:
[   53.008961] CPU: 3 PID: 344 Comm: umount Not tainted 4.3.3 #13
[   53.010213] Hardware name:                  /D54250WYK, BIOS WYLPT10H.86A.0041.2015.0720.1108 07/20/2015
[   53.011505]  ffffffff81cf9a94 ffff880214defc60 ffffffff8131d7cf ffff880214defca8
[   53.012782]  ffff880214defc98 ffffffff81051b5d ffff8800d4f34000 ffff880214d90000
[   53.014088]  ffff880214d90050 0000000000000000 0000000000000001 ffff880214defcf8
[   53.015359] Call Trace:
[   53.016636]  [<ffffffff8131d7cf>] dump_stack+0x44/0x55
[   53.017922]  [<ffffffff81051b5d>] warn_slowpath_common+0x7d/0xb0
[   53.019213]  [<ffffffff81051bd7>] warn_slowpath_fmt+0x47/0x50
[   53.020475]  [<ffffffff81241989>] bdev_read+0xc9/0xd0
[   53.021765]  [<ffffffff8123a7c5>] sector_read+0x45/0x80
[   53.023043]  [<ffffffff8123b30d>] fs_set_vol_flags+0x8d/0xa0
[   53.024325]  [<ffffffff8123b347>] ffsUmountVol+0x27/0x90
[   53.025571]  [<ffffffff81241126>] FsUmountVol+0x36/0x60
[   53.026861]  [<ffffffff8123e7d5>] exfat_put_super+0x25/0x90
[   53.028105]  [<ffffffff81166515>] generic_shutdown_super+0x65/0xe0
[   53.029372]  [<ffffffff81166832>] kill_block_super+0x22/0x70
[   53.030616]  [<ffffffff8123e96b>] exfat_debug_kill_sb+0x3b/0x40
[   53.031906]  [<ffffffff81166b29>] deactivate_locked_super+0x39/0x70
[   53.033156]  [<ffffffff81166bb7>] deactivate_super+0x57/0x60
[   53.034432]  [<ffffffff81181c1a>] cleanup_mnt+0x3a/0x80
[   53.035679]  [<ffffffff81181c9d>] __cleanup_mnt+0xd/0x10
[   53.036953]  [<ffffffff8106b62e>] task_work_run+0x6e/0x90
[   53.038180]  [<ffffffff81002a2e>] prepare_exit_to_usermode+0x9e/0xc0
[   53.039427]  [<ffffffff81002a8c>] syscall_return_slowpath+0x3c/0x100
[   53.040643]  [<ffffffff8193ba67>] int_ret_from_sys_call+0x25/0x8f
[   53.041892] ---[ end trace 30b0ae48bb81ab5d ]---

exdeus7, 2016-06-30:
Same problem here!

dorimanx, 2016-11-21:
Need to see if there is a correct unmount support, as it's never was made for pulling before unmount.
Checks has to be added to auto unmount when device is removed.
I am sure there are devs that can think of something, especially adding support for main line kernel 4.x

ehem, 2017-06-16:
Pulling media before umount happens.  This is never a Good Thing(tm), but filesystems **must** be able to handle this situation without panics.  exfat-nofuse also needs to be able to deal with failing media (which can result in similar situations).

yeruichc, 2018-02-28:
Is this problem solved?

RankHuang, 2018-08-29:
same problem!
any update?