dorimanx/exfat-nofuse #81: Does not compile with Kernel 4.5 (rc1)

alexzeitgeist, 2016-01-30:
Tried to compile (git Nov 10, 2015) with Kernel 4.5-rc1.

> DKMS make.log for exfat-1.2.9+20151110 for kernel 4.5.0-040500rc1-generic (x86_64)
Fri Jan 29 19:12:27 CET 2016
make: Entering directory '/usr/src/linux-headers-4.5.0-040500rc1-generic'
  LD      /var/lib/dkms/exfat/1.2.9+20151110/build/built-in.o
  CC [M]  /var/lib/dkms/exfat/1.2.9+20151110/build/exfat_core.o
  CC [M]  /var/lib/dkms/exfat/1.2.9+20151110/build/exfat_super.o
/var/lib/dkms/exfat/1.2.9+20151110/build/exfat_super.c:1379:2: error: unknown field ‘follow_link’ specified in initializer
  .follow_link = exfat_follow_link,
  ^
/var/lib/dkms/exfat/1.2.9+20151110/build/exfat_super.c:1379:17: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
  .follow_link = exfat_follow_link,
                 ^
/var/lib/dkms/exfat/1.2.9+20151110/build/exfat_super.c:1379:17: note: (near initialization for ‘exfat_symlink_inode_operations.create’)
scripts/Makefile.build:258: recipe for target '/var/lib/dkms/exfat/1.2.9+20151110/build/exfat_super.o' failed
make[1]: *** [/var/lib/dkms/exfat/1.2.9+20151110/build/exfat_super.o] Error 1
Makefile:1391: recipe for target '_module_/var/lib/dkms/exfat/1.2.9+20151110/build' failed
make: *** [_module_/var/lib/dkms/exfat/1.2.9+20151110/build] Error 2
make: Leaving directory '/usr/src/linux-headers-4.5.0-040500rc1-generic'

freswa, 2016-02-04:
Same for me on RC2

ryleyangus, 2016-02-08:
I got the same issue on rc3 as well. I had a go a quick patch after looking at the changes in 4.5-rc1. I haven't tested it fully yet.

`--- exfat_super.4.5.c   2016-02-09 09:54:17.606124077 +1100
+++ exfat_super.c       2016-02-09 09:55:17.809459213 +1100
@@ -1359,13 +1359,7 @@
 /*======================================================================*/
 /*  File Operations                                                     */
 /*======================================================================*/
-#if LINUX_VERSION_CODE >= KERNEL_VERSION(4,5,0)
-static const char *exfat_get_link(struct dentry *dentry, struct inode *inode, void **cookie)
-{
-       struct exfat_inode_info *ei = EXFAT_I(dentry->d_inode);
-       return *cookie = (char *)(ei->target);
-}
-#elif LINUX_VERSION_CODE > KERNEL_VERSION(4,1,0)
+#if LINUX_VERSION_CODE > KERNEL_VERSION(4,1,0)
 static const char *exfat_follow_link(struct dentry *dentry, void **cookie)
 {
        struct exfat_inode_info *ei = EXFAT_I(dentry->d_inode);
@@ -1382,10 +1376,7 @@

 const struct inode_operations exfat_symlink_inode_operations = {
        .readlink    = generic_readlink,
-       #if LINUX_VERSION_CODE < KERNEL_VERSION(4,5,0)
-               .follow_link = exfat_follow_link,
-       #endif
-       .get_link = exfat_get_link,
+       .follow_link = exfat_follow_link,
 };

 static int exfat_file_release(struct inode *inode, struct file *filp)
`

dorimanx, 2016-02-09:
can you please submit pull request, so your name will be in contributors.
Thanks for review. i am stuck with 3.4.110 so i cant help much.

dorimanx, 2016-02-09:
Should be OK now after update by @ryleyangus

closed 2016-02-09 by dorimanx