dorimanx/exfat-nofuse #84: Superblock Changes Prevent Compiling Against Linus's Tree

FireBurn, 2016-05-18:
The superblock changes brought in with the merge window prevent exfat from building

exfat_direct_IO no longer needs the offset and something else has changed too as it then complains about something else once that is removed

FireBurn, 2016-05-18:
CC      fs/exfat/exfat_cache.o
fs/exfat/exfat_core.c: In function ‘ffsMountVol’:
fs/exfat/exfat_core.c:183:2: warning: this ‘if’ clause does not guard... [-Wmisleading-indentation]
  if (sector_read(sb, 0, &tmp_bh, 1) != FFS_SUCCESS)
  ^~
fs/exfat/exfat_core.c:186:3: note: ...this statement, but the latter is misleadingly indented as if it is guarded by the ‘if’
   p_fs->PBR_sector = 0;
   ^~~~
fs/exfat/exfat_super.c: In function ‘exfat_direct_IO’:
fs/exfat/exfat_super.c:1704:6: warning: passing argument 4 of ‘blockdev_direct_IO’ makes pointer from integer without a cast [-Wint-conversion]
      offset, exfat_get_block);
      ^~~~~~
In file included from include/linux/seq_file.h:10:0,
                 from fs/exfat/exfat_super.c:56:
include/linux/fs.h:2817:23: note: expected ‘int (*)(struct inode *, sector_t,  struct buffer_head *, int) {aka int (*)(struct inode *, long unsigned int,  struct buffer_head *, int)}’ but argument is of type ‘loff_t {aka long long int}’
 static inline ssize_t blockdev_direct_IO(struct kiocb *iocb,
                       ^~~~~~~~~~~~~~~~~~
fs/exfat/exfat_super.c:1703:8: error: too many arguments to function ‘blockdev_direct_IO’
  ret = blockdev_direct_IO(iocb, inode, iter,
        ^~~~~~~~~~~~~~~~~~
In file included from include/linux/seq_file.h:10:0,
                 from fs/exfat/exfat_super.c:56:
include/linux/fs.h:2817:23: note: declared here
 static inline ssize_t blockdev_direct_IO(struct kiocb *iocb,
                       ^~~~~~~~~~~~~~~~~~
fs/exfat/exfat_super.c: At top level:
fs/exfat/exfat_super.c:1763:17: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
  .direct_IO   = exfat_direct_IO,
                 ^~~~~~~~~~~~~~~
fs/exfat/exfat_super.c:1763:17: note: (near initialization for ‘exfat_aops.direct_IO’)
cc1: some warnings being treated as errors
scripts/Makefile.build:291: recipe for target 'fs/exfat/exfat_super.o' failed
make[2]: *** [fs/exfat/exfat_super.o] Error 1
make[2]: *** Waiting for unfinished jobs....
scripts/Makefile.build:440: recipe for target 'fs/exfat' failed
make[1]: *** [fs/exfat] Error 2
Makefile:962: recipe for target 'fs' failed
make: *** [fs] Error 2
make: *** Waiting for unfinished jobs....

FireBurn, 2016-05-18:
This patch gets it compiling for me, will need to update the IFDEFs once 4.7-rc1 has been cut

diff --git a/exfat_super.c b/exfat_super.c
index 02d4fd2..ae408be 100644
--- a/exfat_super.c
+++ b/exfat_super.c
@@ -1671,8 +1671,7 @@ static ssize_t exfat_direct_IO(int rw, struct kiocb *iocb,
 static ssize_t exfat_direct_IO(int rw, struct kiocb *iocb,
 					   struct iov_iter *iter, loff_t offset)
 #else /* >= 4.1.x */
-static ssize_t exfat_direct_IO(struct kiocb *iocb,
-					   struct iov_iter *iter, loff_t offset)
+static ssize_t exfat_direct_IO(struct kiocb *iocb, struct iov_iter *iter)
 #endif
 {
 	struct inode *inode = iocb->ki_filp->f_mapping->host;
@@ -1695,13 +1694,13 @@ static ssize_t exfat_direct_IO(struct kiocb *iocb,
 		if (EXFAT_I(inode)->mmu_private < (offset + iov_length(iov, nr_segs)))
 #endif
 #else
-		if (EXFAT_I(inode)->mmu_private < (offset + iov_iter_count(iter)))
+		if (EXFAT_I(inode)->mmu_private < (iov_iter_count(iter)))
 #endif
 			return 0;
 	}
 #if LINUX_VERSION_CODE >= KERNEL_VERSION(4,1,0)
 	ret = blockdev_direct_IO(iocb, inode, iter,
-					offset, exfat_get_block);
+					exfat_get_block);
 #elif LINUX_VERSION_CODE >= KERNEL_VERSION(3,16,0)
 	ret = blockdev_direct_IO(rw, iocb, inode, iter,
 					offset, exfat_get_block);
@@ -1720,7 +1719,7 @@ static ssize_t exfat_direct_IO(struct kiocb *iocb,
 
 #if LINUX_VERSION_CODE >= KERNEL_VERSION(3,16,0)
 	if ((ret < 0) && (rw & WRITE))
-		exfat_write_failed(mapping, offset+iov_iter_count(iter));
+		exfat_write_failed(mapping, iov_iter_count(iter));
 #elif LINUX_VERSION_CODE > KERNEL_VERSION(2,6,34)
 	if ((ret < 0) && (rw & WRITE))
 #ifdef CONFIG_AIO_OPTIMIZATION

FireBurn, 2016-05-18:
This get's it compiling for me, I'll need to sort the IFDEFS once 4.7-rc1 has been cut

`diff --git a/exfat_super.c b/exfat_super.c
index 02d4fd2..ae408be 100644
--- a/exfat_super.c
+++ b/exfat_super.c
@@ -1671,8 +1671,7 @@ static ssize_t exfat_direct_IO(int rw, struct kiocb *iocb,
 static ssize_t exfat_direct_IO(int rw, struct kiocb *iocb,
 					   struct iov_iter *iter, loff_t offset)
 #else /* >= 4.1.x */
-static ssize_t exfat_direct_IO(struct kiocb *iocb,
-					   struct iov_iter *iter, loff_t offset)
+static ssize_t exfat_direct_IO(struct kiocb *iocb, struct iov_iter *iter)
 #endif
 {
 	struct inode *inode = iocb->ki_filp->f_mapping->host;
@@ -1695,13 +1694,13 @@ static ssize_t exfat_direct_IO(struct kiocb *iocb,
 		if (EXFAT_I(inode)->mmu_private < (offset + iov_length(iov, nr_segs)))
 #endif
 #else
-		if (EXFAT_I(inode)->mmu_private < (offset + iov_iter_count(iter)))
+		if (EXFAT_I(inode)->mmu_private < (iov_iter_count(iter)))
 #endif
 			return 0;
 	}
 #if LINUX_VERSION_CODE >= KERNEL_VERSION(4,1,0)
 	ret = blockdev_direct_IO(iocb, inode, iter,
-					offset, exfat_get_block);
+					exfat_get_block);
 #elif LINUX_VERSION_CODE >= KERNEL_VERSION(3,16,0)
 	ret = blockdev_direct_IO(rw, iocb, inode, iter,
 					offset, exfat_get_block);
@@ -1720,7 +1719,7 @@ static ssize_t exfat_direct_IO(struct kiocb *iocb,
 
 #if LINUX_VERSION_CODE >= KERNEL_VERSION(3,16,0)
 	if ((ret < 0) && (rw & WRITE))
-		exfat_write_failed(mapping, offset+iov_iter_count(iter));
+		exfat_write_failed(mapping, iov_iter_count(iter));
 #elif LINUX_VERSION_CODE > KERNEL_VERSION(2,6,34)
 	if ((ret < 0) && (rw & WRITE))
 #ifdef CONFIG_AIO_OPTIMIZATION
`

dorimanx, 2016-11-21:
Can you please file a pull request, so it's can be merged in tree for all.
I can add the patch, but i like to see full credits to author.
Thanks.

dorimanx, 2017-06-19:
was fixed by this commit:
https://github.com/dorimanx/exfat-nofuse/commit/41343168e7eafe58d5fd7021cf7d085c647fe55a
thanks to @FireBurn for the patch.

closed 2017-06-19 by dorimanx