dorimanx/exfat-nofuse #88: d_type

piranna, 2016-08-27:
Add support for `d_type` info so it can be used as a upper filesystem for OverlayFS

dorimanx, 2017-06-19:
closing old requests.

piranna, 2017-06-19:
Could this be open again? I'm still interested on this feature.

dorimanx, 2017-06-19:
No one made any update for that from last year, so i think its will be declined.

piranna, 2017-06-20:
Mostly it's just add the field to a structure and fill it with the file type, it's an easy task. Would you be willing to implement it? If not, would you accept a pull-request?

piranna, 2017-06-20:
Here you can find more info about the d_type field:

http://www.gnu.org/software/libc/manual/html_node/Directory-Entries.html
http://man7.org/linux/man-pages/man3/readdir.3.html

dorimanx, 2017-06-20:
If you can provide pull request with change that stable, and not harming any kernel distribution when building, i will merge it.

piranna, 2017-06-21:
When [reading a file](https://github.com/dorimanx/exfat-nofuse/blob/494505c90090b77120ca04d36154b49d8850c3a6/exfat_super.c#L588-L589) it's being given the type to `dir_emit()`, could it be that this is already being supported? Could we notify the other file types (specially symlinks)?

timofonic, 2017-08-04:
@piranna Are you willing to provide a pull request about it?

piranna, 2017-08-04:
> @piranna Are you willing to provide a pull request about it?

Not too much time lately, but it should be simple. If you want to do it, I can review and give advice :-)