dorimanx/exfat-nofuse #99: no support ftruncate operation support?

lintel, 2016-12-18:
Hi,guys.
Does exfat-nofuse can support linux  fallocate or  posix fallocate operation?

Here is some patches for linux vfat support fallocate:
https://patchwork.kernel.org/patch/1589161/

ehem, 2017-06-16:
Are you interested in the ftruncate() operation?  The fallocate() operation?

These are two distinct operations.  ftruncate() is simply set the size of file X to Y; whereas fallocate() is set the size of file X to Y and ensure space is allocated so subsequent writes _will not_ fail.

dorimanx, 2017-06-19:
I was about to close the issue with "closing old requests."
Lets wait for answer for few days...

lintel, 2017-06-25:
I very sorry for my mistake, 
I found some poblem with samba(**strict allocate = yes**) for exFAT/vFAT system,
because lost fallocate syscall on exFAT/vFAT partitions.

timofonic, 2017-08-04:
@lintel I would say to provide more details and a reproducible test if possible. I'm not a dev, but that's what I heard of them ;)