dorimanx/exfat-nofuse #104: Undefined symbol wait_on_page_writeback_range (Kernel 2.6.31.8)

jefflessard, 2017-02-03:
Hi

I am trying to use your module on an Iomega ix2-200 NAS which runs kernel version 2.6.31.8 (this is the latest firmware available from manufacturer).  At compile time (make), I get a warning because of undefined 'wait_on_page_writeback_range". When trying to load the module using insmod, it fails and I get dmesg error : "exfat: Unknown symbol wait_on_page_writeback_range".

After reviewing kernel source code (from kernel.org) for this specific version, I found that this function is defined into mm/filemap.c but that the symbol is not exported such as the other functions are.

I also observed that in file exfat_super.c you call filemap_fdatawait_range for kernels greater than 2.6.32. The problem is that neither functions are available for kernels version 2.6.31.8.

Do you think this could be workarounded?

Thank you

Jeff

dorimanx, 2017-02-03:
Hmm, export the function in your source,  so module will use it.

jefflessard, 2017-02-03:
I would prefer not to rely on an incomplete kernel. I do not have kernel
sources from the manufacturer nor their patches.

As the function is just not exported into mainstream kernel source, I guess
that anyone running the same version as me will get this problem.

Do you think a fix in exfat-nofuse is possible?

timofonic, 2017-08-04:
@jefflessard Did you fix it?

jefflessard, 2017-08-04:
I have no much time available so I gave up using exfat on this device.