dorimanx/exfat-nofuse #117: Add symlink support

piranna, 2017-06-20:
Althought not part of the spec of ExFAT format, OSX added an extension to FAT filesystems that allow to have symlinks on them, by using special easy to identify ad-hoc files. We could be able to add this functionality too, on a modern UNIX system they are fairly important...

https://groups.google.com/forum/m/#!topic/exfat/vITmn9WcSd4

dorimanx, 2017-06-21:
Exfat is a window file system, linux symlinks will not work as expected. 
You can create windows file links in exfat sdcard, and they will work under windows. 

No need to mix different file system types.

piranna, 2017-06-21:
Can you explain about "windows symlinks"? So far, I only know they are
supported on NTFS, not on ExFAT...

On the other hand, reviewing the code I've find some code to create and
read symlinks using some attributes, but found no documentation about how
they work. Does already exfat-nofuse support symlinks?

piranna, 2017-06-21:
I've review the code and seems `exfat-nofuse` is using [a reserved field](https://github.com/dorimanx/exfat-nofuse/blob/051167ceccbe75d9d77fc4f5356f5a39a01779e8/exfat_api.h#L70) to store a flag to say that file [is in fact a symlink](https://github.com/dorimanx/exfat-nofuse/blob/051167ceccbe75d9d77fc4f5356f5a39a01779e8/exfat_super.h#L148-L149), and I'm not sure if it's doing the same trick with FAT32 filesystems too. Does that means that it support symlinks the UNIX way? In that case, then it would only needs to notify it when [reading a file](https://github.com/dorimanx/exfat-nofuse/blob/494505c90090b77120ca04d36154b49d8850c3a6/exfat_super.c#L588-L589)...

I'm thinking to use it as $HOME for [NodeOS](https://node-os.com) when booting directly from a pendrive.

timofonic, 2017-08-04:
@piranna It seems exfat has many hidden surprises.

@dorimanx Do you have some reply about that? I did believe ExFAT to be a far simpler file system...

piranna, 2017-08-04:
> @piranna It seems exfat has many hidden surprises.

Not ExFAT, but `exfat-nofuse` ;-)