dorimanx/exfat-nofuse #118: Use ATTR_SYSTEM

piranna, 2017-07-03:
`ATTR_SYSTEM` attribute could be used as a lightweight user access control, only allowing to PID 0 (root) access files and folders that has this attribute enabled. This could be enabled by a mount flag, and would be as simple as unsetting the `group` and `other` permissions when enabled.

piranna, 2017-07-04:
I have been thinking, the `ATTR_HIDDEN` would also be used to disable the UNIX `read` permission, similar to how the `ATTR_READONLY` flag is used for the `write` permission. This way, the `ATTR_SYSTEM` would only change the file ownership to root when calling the `getattr()` function (stat), leaving the other ones to the pid of the user that mounted the filesystem.

dorimanx, 2017-07-04:
If you think its can bring improvement for all users, send pull request with all defined and tested.

piranna, 2017-08-05:
According to https://en.wikipedia.org/wiki/Design_of_the_FAT_file_system#Directory_entry, seems several OSes (DR-DOS, OpenDOS, Multiuser-DOS and other similar ones) make use of reserved bytes `0x12-0x15` to store both user IDs and UNIX-like file permissions as a extension to FAT filesystem. I've not been able to find a reference to actual usage of this reserved bytes on Microsoft systems, except that this reserved bytes are also being used by VFAT long names support, but this entries would be easily detected by checking the file attributes, so I think it would be feasable to add support for real POSIX permissions here. What do you think?

piranna, 2017-08-05:
Main problem here with UIDs and GIDs is that they are stored on a single byte, while standard UNIXes use two bytes, so it would need to give a base offset as parameter (Debian/Ubuntu regular users UIDs and GIDs start at 1000 for default user), or a mapping file between both systems.

piranna, 2017-08-05:
ExFAT don't provide space in the same place to implement the DR-DOS POSIX-like permissions, but has a lot of reserved bytes at the end of its structure that could be used for this task, allowing real UIDs, GIDs and permissions. On the other hand, the `ATTR_DEVICE` bit could be used as a replacement for execution bit in a simple way doing the same thing as `ATTR_VOLUME` is being used for symlinks, and being this change compatible with all FAT versions.