dorimanx/exfat-nofuse #136: Unable to compile for 4.18.1 kernel

alphazo, 2018-08-17:
I've been using exfat-nofuse for a long time but something seems to be broken with 4.18.1 kernel:

```
DKMS make.log for exfat-198.01c30ad for kernel 4.18.1-arch1-1-ARCH (x86_64)
Fri Aug 17 22:22:51 CEST 2018
make -C /usr/lib/modules/4.18.1-arch1-1-ARCH/build M=/var/lib/dkms/exfat/198.01c30ad/build modules
make[1] : on entre dans le répertoire « /usr/lib/modules/4.18.1-arch1-1-ARCH/build »
  CC [M]  /var/lib/dkms/exfat/198.01c30ad/build/exfat_core.o
  CC [M]  /var/lib/dkms/exfat/198.01c30ad/build/exfat_super.o
/var/lib/dkms/exfat/198.01c30ad/build/exfat_super.c: Dans la fonction « exfat_fill_inode »:
/var/lib/dkms/exfat/198.01c30ad/build/exfat_super.c:1927:27: error: passage de l'argument 2 de « exfat_time_fat2unix » depuis un type pointeur incompatible [-Werror=incompatible-pointer-typ
es]
  exfat_time_fat2unix(sbi, &inode->i_mtime, &info.ModifyTimestamp);
                           ^~~~~~~~~~~~~~~
/var/lib/dkms/exfat/198.01c30ad/build/exfat_super.c:150:70: note: « struct timespec * » attendu mais l'argument est de type « struct timespec64 * »
 void exfat_time_fat2unix(struct exfat_sb_info *sbi, struct timespec *ts,
                                                     ~~~~~~~~~~~~~~~~~^~
/var/lib/dkms/exfat/198.01c30ad/build/exfat_super.c:1928:27: error: passage de l'argument 2 de « exfat_time_fat2unix » depuis un type pointeur incompatible [-Werror=incompatible-pointer-typ
es]
  exfat_time_fat2unix(sbi, &inode->i_ctime, &info.CreateTimestamp);
                           ^~~~~~~~~~~~~~~
/var/lib/dkms/exfat/198.01c30ad/build/exfat_super.c:150:70: note: « struct timespec * » attendu mais l'argument est de type « struct timespec64 * »
 void exfat_time_fat2unix(struct exfat_sb_info *sbi, struct timespec *ts,
                                                     ~~~~~~~~~~~~~~~~~^~
/var/lib/dkms/exfat/198.01c30ad/build/exfat_super.c:1929:27: error: passage de l'argument 2 de « exfat_time_fat2unix » depuis un type pointeur incompatible [-Werror=incompatible-pointer-typ
es]
  exfat_time_fat2unix(sbi, &inode->i_atime, &info.AccessTimestamp);
                           ^~~~~~~~~~~~~~~
/var/lib/dkms/exfat/198.01c30ad/build/exfat_super.c:150:70: note: « struct timespec * » attendu mais l'argument est de type « struct timespec64 * »
 void exfat_time_fat2unix(struct exfat_sb_info *sbi, struct timespec *ts,

```

alphazo, 2018-08-18:
Thanks @barrybingo. You saved my day since I couldn't copy a bunch of photos from my SD card. I hope this simple fix goes to mainstream soon.

CodeUnit, 2018-09-06:
Here my timespec64 patch:
```diff --git a/exfat_super.c b/exfat_super.c
index 312de36..6c097f5 100644
--- a/exfat_super.c
+++ b/exfat_super.c
@@ -147,8 +147,13 @@ static time_t accum_days_in_year[] = {
 static void _exfat_truncate(struct inode *inode, loff_t old_size);
 
 /* Convert a FAT time/date pair to a UNIX date (seconds since 1 1 70). */
+#if LINUX_VERSION_CODE >= KERNEL_VERSION(4,18,0)
+void exfat_time_fat2unix(struct exfat_sb_info *sbi, struct timespec64 *ts,
+						 DATE_TIME_T *tp)
+#else
 void exfat_time_fat2unix(struct exfat_sb_info *sbi, struct timespec *ts,
 						 DATE_TIME_T *tp)
+#endif
 {
 	time_t year = tp->Year;
 	time_t ld;
@@ -166,8 +171,13 @@ void exfat_time_fat2unix(struct exfat_sb_info *sbi, struct timespec *ts,
 }
 
 /* Convert linear UNIX date to a FAT time/date pair. */
+#if LINUX_VERSION_CODE >= KERNEL_VERSION(4,18,0)
+void exfat_time_unix2fat(struct exfat_sb_info *sbi, struct timespec64 *ts,
+						 DATE_TIME_T *tp)
+#else
 void exfat_time_unix2fat(struct exfat_sb_info *sbi, struct timespec *ts,
 						 DATE_TIME_T *tp)
+#endif
 {
 	time_t second = ts->tv_sec;
 	time_t day, month, year;
```

btaroli, 2018-09-20:
Hoping this makes it into the project soon... having to manually patch each time I upgrade the kernel lately... 4.18.7 now.. :wink:

ddscentral, 2018-10-24:
Same issue with the new kernel 4.19. In addition to the patch, you also need to replace timespec with timespec64 and ktime_get_real_ts with ktime_get_real_ts64 in exfat_oal.c. I'll try to make a proper patch later.

tommyjcarpenter, 2018-12-31:
@barrybingo I was using your local branch for the past several months, but it has failed for me on 4.20

tommyjcarpenter, 2018-12-31:
@ddscentral do you have a proper working patch for 4.20?

ddscentral, 2018-12-31:
I hand-patched the driver on one of my devices because I needed to get the driver working ASAP. I didn't have time to make a proper patch, sorry.

Barrybingo's repo already includes a patch from CodeUnit  (it's missing ifdefs though) to fix timespec64 errors. Your patch fixes exfat_oal.c errors. This should be enough to make exfat work on 4.20.

Be aware though, that to avoid breaking exfat-nofuse for older kernels, you should add ifdefs instead of just patching existing code.

See CodeUnit's patch here: https://github.com/dorimanx/exfat-nofuse/issues/136#issuecomment-419014014

I would also fix Barrybingo's patch since it's missing ifdefs and potentially breaks support for older kernels.

On 12/31/2018 3:02 PM, Tommy Carpenter wrote:

@ddscentral<https://github.com/ddscentral> do you have a proper working patch for 4.20?

tommyjcarpenter, 2018-12-31:
@ddscentral yes I saw @barrybingo has updated so I will go back to using that branch. Hopefully the arch package `exfat-dkms-git` gets fixed, or changed to barry's upstream, because that package has been broken for months.