dorimanx/exfat-nofuse #145: v5.0 patch

CodeUnit, 2019-03-05:
For linux-v5.0 this patch is needed (MS_toSB_macros.patch):
```diff --git a/exfat_core.c b/exfat_core.c
index 143b721..7c4d3c2 100644
--- a/exfat_core.c
+++ b/exfat_core.c
@@ -1757,8 +1757,13 @@ void fs_error(struct super_block *sb)
 
 	if (opts->errors == EXFAT_ERRORS_PANIC)
 		panic("[EXFAT] Filesystem panic from previous error\n");
+#if LINUX_VERSION_CODE >= KERNEL_VERSION(5,0,0)
+        else if ((opts->errors == EXFAT_ERRORS_RO) && !(sb->s_flags & SB_RDONLY)) {
+                sb->s_flags |= SB_RDONLY;
+#else
 	else if ((opts->errors == EXFAT_ERRORS_RO) && !(sb->s_flags & MS_RDONLY)) {
 		sb->s_flags |= MS_RDONLY;
+#endif
 		printk(KERN_ERR "[EXFAT] Filesystem has been set read-only\n");
 	}
 }
diff --git a/exfat_super.c b/exfat_super.c
index 312de36..cee7508 100644
--- a/exfat_super.c
+++ b/exfat_super.c
@@ -2080,7 +2090,11 @@ static void exfat_write_super(struct super_block *sb)
 
 	__set_sb_clean(sb);
 
+#if LINUX_VERSION_CODE >= KERNEL_VERSION(5,0,0)
+	if (!(sb->s_flags & SB_RDONLY))
+#else
 	if (!(sb->s_flags & MS_RDONLY))
+#endif
 		FsSyncVol(sb, 1);
 
 	__unlock_super(sb);
@@ -2136,7 +2150,11 @@ static int exfat_statfs(struct dentry *dentry, struct kstatfs *buf)
 
 static int exfat_remount(struct super_block *sb, int *flags, char *data)
 {
+#if LINUX_VERSION_CODE >= KERNEL_VERSION(5,0,0)
+	*flags |= SB_NODIRATIME;
+#else
 	*flags |= MS_NODIRATIME;
+#endif	
 	return 0;
 }
 
@@ -2489,7 +2507,11 @@ static int exfat_fill_super(struct super_block *sb, void *data, int silent)
 	mutex_init(&sbi->s_lock);
 #endif
 	sb->s_fs_info = sbi;
+#if LINUX_VERSION_CODE >= KERNEL_VERSION(5,0,0)
+	sb->s_flags |= SB_NODIRATIME;
+#else
 	sb->s_flags |= MS_NODIRATIME;
+#endif
 	sb->s_magic = EXFAT_SUPER_MAGIC;
 	sb->s_op = &exfat_sops;
 	sb->s_export_op = &exfat_export_ops;
```

SchrodingerZhu, 2019-03-08:
```
  CC [M]  /build/bbswitch-0.8/bbswitch.o
  CC [M]  /build/source/exfat_core.o
  CC [M]  /build/source/nvidiabl-module.o
/build/source/exfat_core.c: In function 'fs_error':
/build/source/exfat_core.c:1760:64: error: 'MS_RDONLY' undeclared (first use in this function); did you mean 'IS_RDONLY'?
  else if ((opts->errors == EXFAT_ERRORS_RO) && !(sb->s_flags & MS_RDONLY)) {
                                                                ^~~~~~~~~
                                                                IS_RDONLY
/build/source/exfat_core.c:1760:64: note: each undeclared identifier is reported only once for each function it appears in

```

Does this patch fix this? @CodeUnit

CodeUnit, 2019-03-08:
Yes, it works for me.

makefu, 2019-09-02:
@CodeUnit thanks for your efforts? care to create a PR for the issue? i'd love to add your patch to the nixpkgs repository similar to 4.18 and 4.20 support we maintain: https://github.com/NixOS/nixpkgs/blob/master/pkgs/os-specific/linux/exfat/default.nix#L31-L36

AdrianBan, 2019-09-05:
Hi all,

I've forked the project to my github and I've added the patches for 4.18-4.20 and also the patch for >= 5.0.
Please take a look https://github.com/AdrianBan/exfat-nofuse . We can try to continue the project here if you are interested. I'm trying to maintain for myself.

Kind regards,
Adrian

makefu, 2019-09-06:
Hey @AdrianBan thanks for your efforts and your willingness to continue maintain exfat-nofuse. I will prepare an update in nixpkgs.