dorimanx/exfat-nofuse #147: Make Error

nobodywasishere, 2019-07-28:
Neofetch:
```
OS: Manjaro Linux x86_64 
Kernel: 5.1.19-1-MANJARO 
Uptime: 10 mins 
Packages: 1201 (pacman) 
Shell: bash 5.0.7 
Resolution: 1920x1080 
DE: Xfce 
WM: Xfwm4 
WM Theme: Adapta-Maia 
Theme: Matcha-dark-sea [GTK2], Adwaita [GTK3] 
Icons: Paper-Mono-Dark [GTK2], Adwaita [GTK3] 
Terminal: xfce4-terminal 
Terminal Font: Monospace 12 
CPU: AMD Ryzen 5 2600X (12) @ 3.600GHz 
GPU: AMD ATI Radeon RX 470/480/570/570X/580/580X/590 
Memory: 2062MiB / 16036MiB 
```
Make err:
```
]$ make
make -C /lib/modules/5.1.19-1-MANJARO/build M=/home/michael/Dev/exfat-nofuse modules
make[1]: Entering directory '/usr/lib/modules/5.1.19-1-MANJARO/build'
  CC [M]  /home/michael/Dev/exfat-nofuse/exfat_core.o
/home/michael/Dev/exfat-nofuse/exfat_core.c: In function ‘fs_error’:
/home/michael/Dev/exfat-nofuse/exfat_core.c:1760:64: error: ‘MS_RDONLY’ undeclared (first use in this function); did you mean ‘IS_RDONLY’?
 1760 |  else if ((opts->errors == EXFAT_ERRORS_RO) && !(sb->s_flags & MS_RDONLY)) {
      |                                                                ^~~~~~~~~
      |                                                                IS_RDONLY
/home/michael/Dev/exfat-nofuse/exfat_core.c:1760:64: note: each undeclared identifier is reported only once for each function it appears in
make[2]: *** [scripts/Makefile.build:276: /home/michael/Dev/exfat-nofuse/exfat_core.o] Error 1
make[1]: *** [Makefile:1571: _module_/home/michael/Dev/exfat-nofuse] Error 2
make[1]: Leaving directory '/usr/lib/modules/5.1.19-1-MANJARO/build'
make: *** [Makefile:35: all] Error 2
```
Not sure where to go from here. Need to mount a PS Vita over USB via VitaShell

freakynl, 2019-08-21:
Hi,

I found that with kernel 5.0 filesystems should no longer use MS_ but SB_.

Changed MS_ to SB_ in 5 locations,

./exfat_super.c:2083:   if (!(sb->s_flags & MS_RDONLY))
./exfat_core.c:1760:    else if ((opts->errors == EXFAT_ERRORS_RO) && !(sb->s_flags & MS_RDONLY)) {
./exfat_core.c:1761:            sb->s_flags |= MS_RDONLY;
./exfat_super.c:2139:   *flags |= MS_NODIRATIME;
./exfat_super.c:2492:   sb->s_flags |= MS_NODIRATIME;

Have different issues now:

make -C /lib/modules/5.2.9-gentoo/build M=/tmp/ex/exfat-nofuse modules
make[1]: Entering directory '/usr/src/linux-5.2.9-gentoo'
  CC [M]  /tmp/ex/exfat-nofuse/exfat_core.o
  CC [M]  /tmp/ex/exfat-nofuse/exfat_super.o
/tmp/ex/exfat-nofuse/exfat_super.c: In function ‘exfat_fill_inode’:
/tmp/ex/exfat-nofuse/exfat_super.c:1927:27: error: passing argument 2 of ‘exfat_time_fat2unix’ from incompatible pointer type [-Werror=incompatible-pointer-types]
 1927 |  exfat_time_fat2unix(sbi, &inode->i_mtime, &info.ModifyTimestamp);
      |                           ^~~~~~~~~~~~~~~
      |                           |
      |                           struct timespec64 *
/tmp/ex/exfat-nofuse/exfat_super.c:150:70: note: expected ‘struct timespec *’ but argument is of type ‘struct timespec64 *’
  150 | void exfat_time_fat2unix(struct exfat_sb_info *sbi, struct timespec *ts,
      |                                                     ~~~~~~~~~~~~~~~~~^~
/tmp/ex/exfat-nofuse/exfat_super.c:1928:27: error: passing argument 2 of ‘exfat_time_fat2unix’ from incompatible pointer type [-Werror=incompatible-pointer-types]
 1928 |  exfat_time_fat2unix(sbi, &inode->i_ctime, &info.CreateTimestamp);
      |                           ^~~~~~~~~~~~~~~
      |                           |
      |                           struct timespec64 *
/tmp/ex/exfat-nofuse/exfat_super.c:150:70: note: expected ‘struct timespec *’ but argument is of type ‘struct timespec64 *’
  150 | void exfat_time_fat2unix(struct exfat_sb_info *sbi, struct timespec *ts,
      |                                                     ~~~~~~~~~~~~~~~~~^~
/tmp/ex/exfat-nofuse/exfat_super.c:1929:27: error: passing argument 2 of ‘exfat_time_fat2unix’ from incompatible pointer type [-Werror=incompatible-pointer-types]
 1929 |  exfat_time_fat2unix(sbi, &inode->i_atime, &info.AccessTimestamp);
      |                           ^~~~~~~~~~~~~~~
      |                           |
      |                           struct timespec64 *
/tmp/ex/exfat-nofuse/exfat_super.c:150:70: note: expected ‘struct timespec *’ but argument is of type ‘struct timespec64 *’
  150 | void exfat_time_fat2unix(struct exfat_sb_info *sbi, struct timespec *ts,
      |                                                     ~~~~~~~~~~~~~~~~~^~
/tmp/ex/exfat-nofuse/exfat_super.c: In function ‘exfat_write_inode’:
/tmp/ex/exfat-nofuse/exfat_super.c:2010:27: error: passing argument 2 of ‘exfat_time_unix2fat’ from incompatible pointer type [-Werror=incompatible-pointer-types]
 2010 |  exfat_time_unix2fat(sbi, &inode->i_mtime, &info.ModifyTimestamp);
      |                           ^~~~~~~~~~~~~~~
      |                           |
      |                           struct timespec64 *
/tmp/ex/exfat-nofuse/exfat_super.c:169:70: note: expected ‘struct timespec *’ but argument is of type ‘struct timespec64 *’
  169 | void exfat_time_unix2fat(struct exfat_sb_info *sbi, struct timespec *ts,
      |                                                     ~~~~~~~~~~~~~~~~~^~
/tmp/ex/exfat-nofuse/exfat_super.c:2011:27: error: passing argument 2 of ‘exfat_time_unix2fat’ from incompatible pointer type [-Werror=incompatible-pointer-types]
 2011 |  exfat_time_unix2fat(sbi, &inode->i_ctime, &info.CreateTimestamp);
      |                           ^~~~~~~~~~~~~~~
      |                           |
      |                           struct timespec64 *
/tmp/ex/exfat-nofuse/exfat_super.c:169:70: note: expected ‘struct timespec *’ but argument is of type ‘struct timespec64 *’
  169 | void exfat_time_unix2fat(struct exfat_sb_info *sbi, struct timespec *ts,
      |                                                     ~~~~~~~~~~~~~~~~~^~
/tmp/ex/exfat-nofuse/exfat_super.c:2012:27: error: passing argument 2 of ‘exfat_time_unix2fat’ from incompatible pointer type [-Werror=incompatible-pointer-types]
 2012 |  exfat_time_unix2fat(sbi, &inode->i_atime, &info.AccessTimestamp);
      |                           ^~~~~~~~~~~~~~~
      |                           |
      |                           struct timespec64 *
/tmp/ex/exfat-nofuse/exfat_super.c:169:70: note: expected ‘struct timespec *’ but argument is of type ‘struct timespec64 *’
  169 | void exfat_time_unix2fat(struct exfat_sb_info *sbi, struct timespec *ts,
      |                                                     ~~~~~~~~~~~~~~~~~^~
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:279: /tmp/ex/exfat-nofuse/exfat_super.o] Error 1
make[1]: *** [Makefile:1597: _module_/tmp/ex/exfat-nofuse] Error 2
make[1]: Leaving directory '/usr/src/linux-5.2.9-gentoo'
make: *** [Makefile:35: all] Error 2

This is on gentoo against 5.2.9 kernel. More is probably needed.

freakynl, 2019-08-21:
FYI: Came across this (ZFS compile issues vs 5.0 kernel) from which I understood the MS_ -> SB_ swap for filesystems:
https://github.com/zfsonlinux/zfs/issues/8264

DraceWang, 2020-08-05:
you can find your solution in my commit:
https://github.com/DraceWang/exfat-nofuse/commit/7723867b559c40fe6ff445c6455a548d9e171169

closed 2022-04-20 by nobodywasishere