dorimanx/exfat-nofuse #148: Kernel 4.18/4.19/4.20 and >= 5.0 fix

AdrianBan, 2019-09-04:
Hi all,

Because I see this source-code is not maintained anymore, I've forked to my Github and I've added the patch for kernel 4.18/4.19/4.20 and also for 5.0.
Please take a look here: https://github.com/AdrianBan/exfat-nofuse

Kind regards,
Adrian

AdrianBan, 2019-09-05:
Update: added support for Debian DKMS packaging from https://github.com/eugenesan/exfat-nofuse and I've adapted a little bit.

warrenc5, 2019-09-15:
Thank you so much!

re2zero, 2021-11-17:
That's  great job!

EasyNetDev, 2021-11-17:
Anyway, this driver is included in Linux main kernel for a while.

re2zero, 2021-11-17:
yes, but I meet some problems while try to back port it to 4.19, and I don't know how to fix them, because they are  using some new features base on main kernel, that are  fs_context.h and fs_parser.h related.

EasyNetDev, 2021-11-17:
Hi @re2zero,

Indeed, you have right. For such old kernels you need something that is working. For sure my fork is working. I was using on my desktop at that time for long time until kernel 5.4 when Samsung decided to push exFAT to mainline.
Hope you find it right for your project.