dorimanx/exfat-nofuse #152: How to compile on RPi?

josephernest, 2020-12-07:
When doing `make` I get this:

```
pi@raspberrypi:~/exfat-nofuse $ make
make -C /lib/modules/5.4.79-v7l+/build M=/home/pi/exfat-nofuse modules
make[1]: Entering directory '/usr/src/linux-headers-5.4.79-v7l+'
  CC [M]  /home/pi/exfat-nofuse/exfat_core.o
/home/pi/exfat-nofuse/exfat_core.c: In function ‘fs_error’:
/home/pi/exfat-nofuse/exfat_core.c:1760:64: error: ‘MS_RDONLY’ undeclared (first use in this function); did you mean ‘IS_RDONLY’?
  else if ((opts->errors == EXFAT_ERRORS_RO) && !(sb->s_flags & MS_RDONLY)) {
                                                                ^~~~~~~~~
                                                                IS_RDONLY
/home/pi/exfat-nofuse/exfat_core.c:1760:64: note: each undeclared identifier is reported only once for each function it appears in
make[2]: *** [scripts/Makefile.build:266: /home/pi/exfat-nofuse/exfat_core.o] Error 1
make[1]: *** [Makefile:1732: /home/pi/exfat-nofuse] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.4.79-v7l+'
make: *** [Makefile:35: all] Error 2
```

How do you install it?

Civil, 2020-12-07:
Starting from 5.4, exfat driver is included in the kernel (it should be in staging). It's in general better to use in-kernel one at this stage. Or if for some reason you really want to install a non-mainline driver, there were some forks of this project that made it compatible with more modern kernels (this repo had no commits for 2+ years, so it won't work out of box with modern kernels)

josephernest, 2020-12-07:
Thanks @civil. It seems that my current kernel is built *without* this module enabled. (Raspberry Pi, Linux version 5.4.79-v7l+)

I tried `modprobe exfat` without success. Is there a way to install it without rebuilding the whole kernel?

Civil, 2020-12-07:
Well, you haven't compiled (this module won't compile for newer kernels). I'd maybe suggest to ask for the working exfat driver on a forum or resoruce for the distro that you are using on raspberry pi.

If you are using raspbian, maybe that's the forum thread that you should look into: https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=275453

TLDR; yes and no. You need to follow parts of the process, enable in-kernel exfat support as module and depends on what was missing, you might only need to configure kernel and recompile modules.