dorimanx/exfat-nofuse #153: Not building against kernel 5.10

daedalus, 2021-02-03:
Sorry for the long post, this is my build-log:

 `
uname -a
Linux wintermute 5.10.0-2-amd64 #1 SMP Debian 5.10.9-1 (2021-01-20) x86_64 GNU/Linux
`

```
dclavijo@wintermute:~/exfat-nofuse$ make -j 8
make -C /lib/modules/5.10.0-2-amd64/build M=/home/dclavijo/exfat-nofuse modules
make[1]: Entering directory '/usr/src/linux-headers-5.10.0-2-amd64'
  CC [M]  /home/dclavijo/exfat-nofuse/exfat_core.o
  CC [M]  /home/dclavijo/exfat-nofuse/exfat_super.o
  CC [M]  /home/dclavijo/exfat-nofuse/exfat_oal.o
  CC [M]  /home/dclavijo/exfat-nofuse/exfat_upcase.o
/home/dclavijo/exfat-nofuse/exfat_oal.c:124:8: error: unknown type name ‘time_t’
  124 | static time_t accum_days_in_year[] = {
      |        ^~~~~~
/home/dclavijo/exfat-nofuse/exfat_oal.c: In function ‘tm_current’:
/home/dclavijo/exfat-nofuse/exfat_oal.c:131:18: error: storage size of ‘ts’ isn’t known
  131 |  struct timespec ts;
      |                  ^~
/home/dclavijo/exfat-nofuse/exfat_oal.c:132:2: error: unknown type name ‘time_t’; did you mean ‘ktime_t’?
  132 |  time_t second, day, leap_day, month, year;
      |  ^~~~~~
      |  ktime_t
/home/dclavijo/exfat-nofuse/exfat_oal.c:137:2: error: implicit declaration of function ‘ktime_get_real_ts’; did you mean ‘ktime_get_real_ns’? [-Werror=implicit-function-declaration]
  137 |  ktime_get_real_ts(&ts);
      |  ^~~~~~~~~~~~~~~~~
      |  ktime_get_real_ns
/home/dclavijo/exfat-nofuse/exfat_oal.c:131:18: warning: unused variable ‘ts’ [-Wunused-variable]
  131 |  struct timespec ts;
      |                  ^~
cc1: some warnings being treated as errors
make[3]: *** [/usr/src/linux-headers-5.10.0-2-common/scripts/Makefile.build:284: /home/dclavijo/exfat-nofuse/exfat_oal.o] Error 1
make[3]: *** Waiting for unfinished jobs....
/home/dclavijo/exfat-nofuse/exfat_super.c:142:8: error: unknown type name ‘time_t’
  142 | static time_t accum_days_in_year[] = {
      |        ^~~~~~
/home/dclavijo/exfat-nofuse/exfat_super.c: In function ‘exfat_time_fat2unix’:
/home/dclavijo/exfat-nofuse/exfat_super.c:153:2: error: unknown type name ‘time_t’; did you mean ‘ktime_t’?
  153 |  time_t year = tp->Year;
      |  ^~~~~~
      |  ktime_t
/home/dclavijo/exfat-nofuse/exfat_super.c:154:2: error: unknown type name ‘time_t’; did you mean ‘ktime_t’?
  154 |  time_t ld;
      |  ^~~~~~
      |  ktime_t
/home/dclavijo/exfat-nofuse/exfat_super.c:161:4: error: invalid use of undefined type ‘struct timespec’
  161 |  ts->tv_sec =  tp->Second  + tp->Minute * SECS_PER_MIN
      |    ^~
/home/dclavijo/exfat-nofuse/exfat_super.c:165:4: error: invalid use of undefined type ‘struct timespec’
  165 |  ts->tv_nsec = 0;
      |    ^~
/home/dclavijo/exfat-nofuse/exfat_super.c: In function ‘exfat_time_unix2fat’:
/home/dclavijo/exfat-nofuse/exfat_super.c:172:2: error: unknown type name ‘time_t’; did you mean ‘ktime_t’?
  172 |  time_t second = ts->tv_sec;
      |  ^~~~~~
      |  ktime_t
/home/dclavijo/exfat-nofuse/exfat_super.c:172:20: error: invalid use of undefined type ‘struct timespec’
  172 |  time_t second = ts->tv_sec;
      |                    ^~
/home/dclavijo/exfat-nofuse/exfat_super.c:173:2: error: unknown type name ‘time_t’; did you mean ‘ktime_t’?
  173 |  time_t day, month, year;
      |  ^~~~~~
      |  ktime_t
/home/dclavijo/exfat-nofuse/exfat_super.c:174:2: error: unknown type name ‘time_t’; did you mean ‘ktime_t’?
  174 |  time_t ld;
      |  ^~~~~~
      |  ktime_t
[... 175 lines cut ...]
  169 | void exfat_time_unix2fat(struct exfat_sb_info *sbi, struct timespec *ts,
      |                                                     ~~~~~~~~~~~~~~~~~^~
/home/dclavijo/exfat-nofuse/exfat_super.c:2011:27: error: passing argument 2 of ‘exfat_time_unix2fat’ from incompatible pointer type [-Werror=incompatible-pointer-types]
 2011 |  exfat_time_unix2fat(sbi, &inode->i_ctime, &info.CreateTimestamp);
      |                           ^~~~~~~~~~~~~~~
      |                           |
      |                           struct timespec64 *
/home/dclavijo/exfat-nofuse/exfat_super.c:169:70: note: expected ‘struct timespec *’ but argument is of type ‘struct timespec64 *’
  169 | void exfat_time_unix2fat(struct exfat_sb_info *sbi, struct timespec *ts,
      |                                                     ~~~~~~~~~~~~~~~~~^~
/home/dclavijo/exfat-nofuse/exfat_super.c:2012:27: error: passing argument 2 of ‘exfat_time_unix2fat’ from incompatible pointer type [-Werror=incompatible-pointer-types]
 2012 |  exfat_time_unix2fat(sbi, &inode->i_atime, &info.AccessTimestamp);
      |                           ^~~~~~~~~~~~~~~
      |                           |
      |                           struct timespec64 *
/home/dclavijo/exfat-nofuse/exfat_super.c:169:70: note: expected ‘struct timespec *’ but argument is of type ‘struct timespec64 *’
  169 | void exfat_time_unix2fat(struct exfat_sb_info *sbi, struct timespec *ts,
      |                                                     ~~~~~~~~~~~~~~~~~^~
cc1: some warnings being treated as errors
make[3]: *** [/usr/src/linux-headers-5.10.0-2-common/scripts/Makefile.build:284: /home/dclavijo/exfat-nofuse/exfat_super.o] Error 1
make[2]: *** [/usr/src/linux-headers-5.10.0-2-common/Makefile:1817: /home/dclavijo/exfat-nofuse] Error 2
make[1]: *** [/usr/src/linux-headers-5.10.0-2-common/Makefile:185: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.10.0-2-amd64'
make: *** [Makefile:35: all] Error 2
```

Civil, 2021-02-04:
You should probably use in-kernel driver (it was included AFAIR since 5.4 and reworked a lot in 5.7). You probably can find it in staging tree.

It is based on a newer version of samsung code, while this repo is long time inactive (no commits for about 4 years or so)

daedalus, 2021-02-04:
I will try it, thanks

closed 2021-02-04 by daedalus